EdjeElectronics/TensorFlow-Object-Detection-on-the-Raspberry-Pi #2: Step 6: python3 Object_detection_picamera.py

otieno-o, 2018-06-18:
Please note I followed the tutorial step by step apart from ignoring step 3 'git checkout v1.8.0', it resulted to deletion of previously downloaded files as highlighted in issue 1. Again very grateful to your efforts @EdjeElectronics . 
Running the command 'python3 Object_detection_picamera.py ' generated the following error. 
[image] What would be the best foot forward? Thanks

Some notes:
(Not sure if relevant) On checking the files (not sure if what I didnt was correct) i noticed there is no file with 'string_int_label_map_pb2' but found 'string_int_label_map.proto'.

EdjeElectronics, 2018-06-18:
Shoot! I forgot to put a critical step in the instructions. All the protobuf files need to be converted from .proto to _pb2.py . I'll look at it, add a step, and update when it's fixed!

EdjeElectronics, 2018-06-18:
Okay, I added instructions in Step 5 to issue a "protoc" command that converts all the .proto files into _pb2.py files. Can you try issuing the following commands and then seeing if Object_detection_picamera.py works?

cd /home/pi/tensorflow1/models/research
protoc object_detection/protos/*.proto --python_out=.

otieno-o, 2018-06-19:
Works like magic..Thanks

closed 2018-06-19 by otieno-o