EdjeElectronics/TensorFlow-Object-Detection-on-the-Raspberry-Pi #8: Slower inference time on modified models

eduagarcia, 2018-07-19:
I trained 2 different models on top of the ssdlite_mobilenet_v2_coco from the tensorflow object api model zoo, but I get different speeds on each model:

ssdlite_mobilenet_v2_coco (90 classes) = 1.3 fps
retrained_ssdlite_1 (2 classes) = 0.7 fps
retrained_ssdlite_2 (3 classes) = 0.8 fps

My trained models are experimenting an drop of almost half the speed of the original model

Is anybody having the same issue?
Am I missing some step to make the model efficient at inference time?

eduagarcia, 2018-07-19:
Nvm, the problem was that I was creating the frozen graph in my personal computer instead of doing directly on the raspberry, I suppose that the tensorflow api makes some kind of optimization according to the the device

closed 2018-07-19 by eduagarcia

roydekleijn, 2018-09-26:
experiencing the same

So the learning also needs to take place on the raspberry pi?

Wouldn't that take ages ? It already takes hours on my mac, with only 6 classes.

Please let me know your experience

eduagarcia, 2018-10-05:
You only export inference graph (.pb) on the raspberry pi

When you finish the training on your mac it generates an model.cpkt-XXXX file right?

Transfer an copy and generate the .pb file on the raspberry pi.

Example
```bash
#Inside the raspberry pi
python export_inference_graph.py --input_type image_tensor --pipeline_config_path training/faster_rcnn_inception_v2_pets.config --trained_checkpoint_prefix training/model.ckpt-XXXX --output_directory inference_graph
```

Twikkxx, 2019-03-07:
hello, How did you export the inference graph on the raspberry pi ? I am doing a project on object detection using raspberry pi. As I tried to import the inference graph in the pi just like what you said I encountered this error.

Traceback (most recent call last):
  File "export_inference_graph.py", line 96, in <module>
    from object_detection import exporter
  File "/home/pi/tensorflow2/models/research/object_detection/exporter.py", line 20, in <module>
    from tensorflow.contrib.quantize.python import graph_matcher
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/contrib/__init__.py", line 38, in <module>
    from tensorflow.contrib import cloud
ImportError: cannot import name 'cloud'
**python3 export_inference_graph.py --input_type image_tensor --pipeline_config_path training/ssd_mobilenet_v2_coco.config. --trained_checkpoint_prefix training/model.ckpt-34491 --output_directory inference_graph
Traceback (most recent call last):
  File "export_inference_graph.py", line 96, in <module>
    from object_detection import exporter
  File "/home/pi/tensorflow2/models/research/object_detection/exporter.py", line 20, in <module>
    from tensorflow.contrib.quantize.python import graph_matcher
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/contrib/__init__.py", line 38, in <module>
    from tensorflow.contrib import cloud
ImportError: cannot import name 'cloud'
**