EdjeElectronics/TensorFlow-Object-Detection-on-the-Raspberry-Pi #15: Detect human only and trigger gpio

mrbensonchong, 2018-09-23:
How can I modify the code to make the camera detect human only. And it can trigger gpio when the number of detected human goes to certain number such as 5?

ISGIND, 2019-01-24:
Did you find a solution?? I have to detect only cars, so if you find the answer for use only one class... pls support me

Twikkxx, 2019-02-20:
Hi did you find a solution for this? I'am working on this kind of project, if the model can detect an object it will trigger the pi gpio

mrbensonchong, 2019-02-21:
I can't find the solution for that. finally i use Google Vision Kit to do
that. the build-in program is good and you can easily amend the code to
trigger the gpio. If you would like to know more about how i do that, i can
share with you the video. thx
Benson

JobinC, 2019-03-06:
Could you share the video for that please @mrbensonchong

servetcoskun, 2019-04-01:
@ISGIND @mrbensonchong you can edit the `mscoco_label_map.pbtxt`file or make a copy and call it something else, then you can change the path in the code to your label file `PATH_TO_LABELS = os.path.join(CWD_PATH, 'data', 'mscoc_label_map_cars.pbtxt')` you can also change number of labels depending on how many you have in your .pbtxt file `NUM_CLASSES = 10`. I got an error after some fiddling around where it will classify the whole image, not sure if its from editing the files and classes variable, but it works nonetheless.