EdjeElectronics/TensorFlow-Object-Detection-on-the-Raspberry-Pi #17: tensorflow-1.10.0-cp35-none-linux_armv7l.whl is not a supported wheel on this platform. Storing debug log for failure in /root/.pip/pip.log

Devraj004, 2018-10-01:
Hi,
This project is really interesting but some how i am struggling from last 2 days and getting this error every time. 
tensorflow-1.10.0-cp35-none-linux_armv7l.whl is not a supported wheel on this platform.
Storing debug log for failure in /root/.pip/pip.log

My raspberry is running on version  "jessie desktop".

Please help me to resolve it. 

Regards
Devraj

EdjeElectronics, 2018-10-03:
You can now install TensorFlow just by issuing these commands in the Pi terminal:

```
sudo apt-get install libatlas-base-dev
pip3 install tensorflow
```

Will you try that and see if it works?

phdaerospace, 2019-04-25:
i had the same problem as Devraj004. I tried your fix Edje but i get this
pi@raspberrypi:~/tf $ pip3 install tensorflow
Downloading/unpacking tensorflow
  Could not find any downloads that satisfy the requirement tensorflow
Cleaning up...
No distributions at all found for tensorflow
Storing debug log for failure in /home/pi/.pip/pip.log
sudo apt install libatlas worked but the install tensorflow on the second line didnt

phdaerospace, 2019-04-25:
Well i found that the tensor flow for ARM has all the whl files for Python 3.5 and i was running python 3.4. I just upgraded the Pyhton to 3.5.2 and tf started installing.