EdjeElectronics/TensorFlow-Object-Detection-on-the-Raspberry-Pi #18: ImportError: libImath-2_2.so.12: cannot open shared object file: No such file or directory

osianSmith, 2018-10-17:
Heya
I'm trying to run the command "python3 Object_detection_picamera.py " and I keep getting the this stack being returned:
```
Traceback (most recent call last):
  File "Object_detection_picamera.py", line 23, in <module>
    import cv2
  File "/home/pi/.local/lib/python3.5/site-packages/cv2/__init__.py", line 3, in <module>
    from .cv2 import *
ImportError: libImath-2_2.so.12: cannot open shared object file: No such file or directory

```

I have followed all the instructions and just wondering what is the problem - have I missed something or is there something else that I need to install? 

Thank you 

Osian

Samitha156, 2018-10-26:
Hey did  you found any solution for this ? maybe the problem is version of raspbian . Is that?

rribani, 2018-10-29:
Install the following packages with apt-get:

libilmbase-dev
libopenexr-dev
libgstreamer1.0-dev

Worked for me!

daverooneyca, 2019-01-02:
I have Raspbian Lite with no UI installed on a Pi 3A+. I followed @rribani's instructions above which got me past the error in this issue, but I encountered a further problem when the script couldn't find the picamera Python module. I installed that using `sudo apt-get install python3-picamera`.

That allowed the script to work, but it killed itself after a minute or so with a std::bad_alloc exception, likely because it couldn't create a window to show the image from the camera.

Now I'm loading a minimal PIXEL desktop... stay tuned.

osianSmith, 2019-01-26:
Hi all
Sorry for my slow reply. I haven't been able to any solutions out due to a change of requirements of my project. Sorry for my slow reply and thank you all for your help :)

closed 2019-01-26 by osianSmith

agneevguin, 2019-10-25:
This is still an open issue. The above apt-gets didnt solve the problem in Raspberry Pi 4.

BotOreo, 2019-11-05:
Hello there. I too encountered the same problem.

I have tried the above apt-get, and it seems they have already the newest version. 

I still encountered the problem after the apt-get command.

I still got the error : - 

ImportError: libImath-2_2.so.12: cannot open shared object file: No such file or directory.

justinmiller24, 2020-01-02:
Hello, I encountered the same error. Install was on a brand new Raspberry Pi 4b with the Raspbian Lite OS.

`ImportError: libImath-2_2.so.12: cannot open shared object file: No such file or directory.`

I traced the dependencies and found that "libImath" is part of the libilmbase23 package, so this fixed the problem:

`sudo apt install libilmbase23`

However, that led to a second problem:

`ImportError: libIlmImf-2_2.so.22: cannot open shared object file: No such file or directory`

Again, tracing the dependencies led to "libIlmImf" being part of the libopenexr package. This was only available in a dev package, so the fixed the second problem:

`sudo apt install libopenexr-dev`

And boom! We are up and running. Hope this helps.

amirtabai, 2020-03-23:
> Hello, I encountered the same error. Install was on a brand new Raspberry Pi 4b with the Raspbian Lite OS.
>
> `ImportError: libImath-2_2.so.12: cannot open shared object file: No such file or [...]

This solved the issue for me!

Lizhiteng, 2020-11-13:
this is no libilmbase23, how to fix it

EdjeElectronics, 2021-03-25:
Thanks a bunch @rribani and @justinmiller24! I just ran in to this issue while trying to set up OpenCV-Python on a Raspberry Pi Zero W, and installing the libilmbase-dev, libopenexr-dev, and libgstreamer1.0-dev packages resolved it. Cheers!

Vishaan24, 2021-06-08:
For me after installing `sudo apt install libopenexr-dev`, I installed another library which is missing `sudo apt-get install -y libgtk-3-dev`

Darksoulz15, 2021-07-08:
> Hello, I encountered the same error. Install was on a brand new Raspberry Pi 4b with the Raspbian Lite OS.
>
> `ImportError: libImath-2_2.so.12: cannot open shared object file: No such file or [...]

this solved the issue

Cankar001, 2023-04-02:
I tried to install all mentioned packages but none of these worked for me. I am using python 3.7 and the error occurs in the mediapipe library, not in Opencv

jirivchi1, 2023-04-17:
> I tried to install all mentioned packages but none of these worked for me. I am using python 3.7 and the error occurs in the mediapipe library, not in Opencv

hey, I am having the same problem and also I am using python 3.7.12. Could you tell if you solved the problem? 
thanks in advence