EdjeElectronics/TensorFlow-Object-Detection-on-the-Raspberry-Pi #22: ImportError: numpy.core.multiarray failed to import

wozwas, 2018-12-14:
Followed guide and everything looks good, but when I run python3 Object_detection_picamera.py I get the following:

ImportError: numpy.core.multiarray failed to import
Traceback (most recent call last):
  File "Object_detection_picamera.py", line 23, in <module>
    import cv2
  File "/home/pi/.local/lib/python3.5/site-packages/cv2/__init__.py", line 3, in <module>
    from .cv2 import *
ImportError: numpy.core.multiarray failed to import

akhzarna, 2019-05-17:
I am also searching for the same issue, If you have fixed the issue please guide me

Rambo-7, 2019-07-29:
I am also searching for the same issue, If you have fixed the issue please guide me too

elektronika-ba, 2019-07-29:
Yep, me too.

Rambo-7, 2019-08-02:
Could some masterminds help us ?😭

Rambo-7, 2019-08-04:
I have solved this problem. Because there are some different editions of numpy in Pi. So uninstall other editions ,just keep one edition only.

edgarscr, 2019-11-09:
> I have solved this problem. Because there are some different editions of numpy in Pi. So uninstall other editions ,just keep one edition only.

can u help me to fix?