EdjeElectronics/TensorFlow-Object-Detection-on-the-Raspberry-Pi #30: File "Object_detection_picamera.py", line 137, in <module> e

VictorTagayun, 2019-01-20:
```
pi@raspberrypi:~/tensorflow1/models/research/object_detection $ python3 Object_detection_picamera.py
Traceback (most recent call last):
  File "Object_detection_picamera.py", line 137, in <module>
    frame.setflags(write=1)
ValueError: cannot set WRITEABLE flag to True of this array
```

EdjeElectronics, 2019-01-21:
It looks like several people have been getting this error. I'm not sure why it happens to some people but not to others. It may be an issue with having different versions of the Picamera or numpy library.

For now, it can be solved by adding the following code directly after line 136.
```
frame = np.copy(frame)
```
However, this is a fairly large computational step that will slightly decrease the FPS, so I'd prefer not to add it. Can anyone help me figure out the root cause of this "ValueError: cannot set WRITEABLE flag to True of this array"? Why is it occurring on some people's devices but not on others?

EdjeElectronics, 2019-01-21:
For now, I have changed line 136 in Object_detection_picamera.py from
```
frame = frame1.array
```
to
```
frame = np.copy(frame1.array)
```

JanLin, 2019-01-23:
The proposed change worked. Thanks!

To help understand the problem here is the version I am using:
numpy==1.16.0
picamera==1.13

If you let me know which version you have used I can test with a different version to see if the problem goes away.

350SBC, 2019-01-24:
This error might be permissions related. The error only occurs if i run the script under the regular user. If i run as root it works.

VictorTagayun, 2019-01-24:
will try later with sudo...
is that correct?

350SBC, 2019-01-24:
> will try later with sudo...
> is that correct?

The fix Evan (EdjeElectronics) implemented works, just re-download the Object_detection_picamera.py 
or edit line 136 as stated above. 

As for your question ...
(security warning!!!! this was just for testing) 
Since I do not have a monitor on my pi;I set a root password and enabled root access in the sshd_config. This way when i shell to the pi the xsession will be owned by root. 

If you have monitor on the pi sudo should work if it complains about  **X11 connection rejected because of wrong authentication.** Then you may need to issue and xhost + to allow root access to your users display.  
(security warning!!!! this was just for testing)

JanLin, 2019-01-24:
Tried root as suggested by adding sudo ... but still same error without the fix. As regular user using fix it works but as was noted framerate is around 0.6.

VictorTagayun, 2019-01-28:
works for me on rpi3b+, framerate is 1.1 - 1.4 fps, ave is 1.25.

VictorTagayun, 2019-01-29:
fps dropped to 0.8 - 1.0 when I increased the screen resolution to 1600 x 1080.
Previous fps  1.1 - 1.4 fps is when screen resolution was 800 x 480 when using 5in LCD screen.
How to reduce the camera resolution to fit in to 800 x 480 or less?

closed 2019-01-31 by VictorTagayun

myworldbox, 2019-12-27:
> For now, I have changed line 136 in Object_detection_picamera.py from
> 
> ```
> frame = frame1.array
> ```
> 
> to
> 
> ```
> frame = np.copy(frame1.array)
> ```

Bro, you save my project, I love you!!!!!!! :DDDDDDD

yeroc-sebrof, 2021-10-12:
Bumping this for the Pi 4B, running as root still getting this ValueError. The np.copy() function takes up a fair chunk of real-time processing so I'd really like to set writable. I've tested all the way back from 1.13 to 1.06 and this issue is present for all versions that had PiRGBAnalysis. Should I be using something other than PiRGBAnalysis for this? Custom write format or even just YUV format perhaps?