EdjeElectronics/TensorFlow-Object-Detection-on-the-Raspberry-Pi #34: ImportError: libgstbase-1.0.so.0: cannot open shared object file: No such file or directory

ozett, 2019-02-24:
i did everything as in dokumentation.

if i run object_detection_picamera.py i got this error.

```ImportError: libgstbase-1.0.so.0: cannot open shared object file: No such file or directory```

any hint what is missing?

thanks a lot

ozett, 2019-02-24:
[image]

import cv2 ?

ozett, 2019-02-24:
libgstreamer1.0-0  was missing? i did:
```apt-get install libgstreamer1.0-0```

now run the script again, got memory error. but seems to be running...
i am headless, so i will try with gui now..

```
root@raspimax7219:/usr/src/tensorflow1/models/research/object_detection# python3 Object_detection_picamera.py
2019-02-24 08:39:04.083671: W tensorflow/core/framework/allocator.cc:101] Allocation of 11059200 exceeds 10% of system memory.
2019-02-24 08:39:04.144215: W tensorflow/core/framework/allocator.cc:101] Allocation of 11059200 exceeds 10% of system memory.
2019-02-24 08:39:04.327620: W tensorflow/core/framework/allocator.cc:101] Allocation of 8640000 exceeds 10% of system memory.
2019-02-24 08:39:04.436385: W tensorflow/core/framework/allocator.cc:101] Allocation of 3240000 exceeds 10% of system memory.
2019-02-24 08:39:04.466177: W tensorflow/core/framework/allocator.cc:101] Allocation of 3240000 exceeds 10% of system memory.
2019-02-24 08:39:04.537811: W tensorflow/core/framework/allocator.cc:101] Allocation of 3240000 exceeds 10% of system memory.
```

ozett, 2019-02-24:
i got it running... wonderfull
😸 😺 

[image]

i installed xrdp to have a remote-gui

edgarscr, 2019-11-09:
> i got it running... wonderfull
>  
> 
> [image]
> 
> i installed xrdp to have a remote-gui

but the reason of the bug its the memory or what? u cnat run on the same raspberry? im trying on raspberry pi 4 2b