EdjeElectronics/TensorFlow-Object-Detection-on-the-Raspberry-Pi #35: headless object_detection.py possible?

ozett, 2019-02-24:
hi,
do you think your script could run on my (everbodies?) headless raspberry-pi?
and save a image-snapshot if an object is detected?

any ideas how to realize that (sorry, i am no coder...)..

thanks, if you find time to answer...
tozett

alexharnozd, 2019-04-02:
I m having a problem with headless.
X11 problem the first. When it done with x server problem. 
It seems its error with cant connect with X server upon running final python3 script start up for Ocject detection

wanghetongtt, 2019-05-29:
> I m having a problem with headless.
> X11 problem the first. When it done with x server problem.
> It seems its error with cant connect with X server upon running final python3 script start up for Ocject detection

same here , did you solve it

ozett, 2019-05-30:
try to connect with mobaterm from a windows workstation
https://mobaxterm.mobatek.net/
to the raspberry.

mobaterm has an x-server included. this works.
or you change the code and disable showing images. (no need for x-server).
instead you could save the images or code something else to alert you. 

rgards,
tozett