EdjeElectronics/TensorFlow-Object-Detection-on-the-Raspberry-Pi #36: cat-pet detector.py error: frame.setFlags

ozett, 2019-02-24:
i wanted to try cat-detection,
(object-detection already running fine)
disabled twilio in the script, but got this error:

```
root@raspimax7219:/usr/src/tensorflow1/models/research/object_detection# python3 Pet_detector.py
Traceback (most recent call last):
  File "Pet_detector.py", line 275, in <module>
    frame.setflags(write=1)
ValueError: cannot set WRITEABLE flag to True of this array
root@raspimax7219:/usr/src/tensorflow1/models/research/object_detection#
```

may you can help me out, shorthand? 
dont understand the array-constrution...but you?
what to do here?

thanks,
tozett

ozett, 2019-02-24:
i would have been to simple.
i disabled the line, the script run firsthand, than trow an error:

[image]

ozett, 2019-02-24:
i compared the object-detector.py with the cat-detector.py and 
changed the pet.py without real understanding at one line according to the object.py
but no it runs..i am waiting now for the cat to come and see if it works..

[image]

EdjeElectronics, 2019-03-16:
Hi @ozett , glad you were able to get it working. Issue #30 discusses how to deal with the "ValueError: cannot set WRITEABLE flag to True of this array" error.

closed 2019-03-16 by EdjeElectronics