EdjeElectronics/TensorFlow-Object-Detection-on-the-Raspberry-Pi #38: "pip3 install opencv-python" is failed

leemgs, 2019-02-27:
It's strange. 
The "pip3 install opencv-python" command generates a error message as following:

```bash
pi@raspberrypi:~ $
pi@raspberrypi:~ $ pip3 install opencv-python
Traceback (most recent call last):
  File "/usr/bin/pip3", line 9, in <module>
    from pip import main
ImportError: cannot import name 'main'
pi@raspberrypi:~ $

```

leemgs, 2019-02-27:
I have solved the below recipes. 

## Method 1: 
If you use latest version of pip3, you need to install opencv-python "python3 -m pip ..." style as following:

```bash
$ python3 -m pip install opencv-python
```

	
## Method 2:
Try to use "sudo" command.

```
pi@raspberrypi:~ $ sudo pip3 install opencv-python
Collecting opencv-python
  Downloading https://www.piwheels.org/simple/opencv-python/opencv_python-3.4.4.19-cp35-cp35m-linux_armv7l.whl (7.4MB)
    100% |################################| 7.4MB 26kB/s
Requirement already satisfied: numpy>=1.12.1 in /usr/local/lib/python3.5/dist-packages (from opencv-python)
Installing collected packages: opencv-python
Successfully installed opencv-python-3.4.4.19
pi@raspberrypi:~ $
```

EdjeElectronics, 2019-03-16:
Thanks for updating with your solution! Glad you were able to get it working.

closed 2019-03-16 by EdjeElectronics