EdjeElectronics/TensorFlow-Object-Detection-on-the-Raspberry-Pi #42: BGR or RGB?

rafitzadik, 2019-04-01:
Thank you for the great tutorial!
I have a question: your code gets the image in BGR (for both the picamera and USB). I thought that the tensorflow pre-built models expect RGB. 

Don't you need to convert (e.g. frame= frame[:, :, [2, 1, 0]]) ?

EdjeElectronics, 2019-04-16:
Hi @rafitzadik , can you point me to where you saw that TensorFlow models expect images in RGB format? If you do, I will try changing converting them and see how it effects the results.

rafitzadik, 2019-04-16:
Thanks for taking a look at this and responding quickly!
Take a look at all of the tf.io.decode_* functions, e.g.
https://www.tensorflow.org/api_docs/python/tf/io/decode_bmp
Also, less directly, at
https://github.com/mystic123/tensorflow-yolo-v3/issues/2
Having said that, I believe that if you fine-tune your own model then this
doesn't matter - just feed it the same way in training and in inference.

Thanks again,

rafitzadik

rafitzadik, 2019-04-17:
I should clarify:
> Having said that, I believe that if you fine-tune your own model then this
> doesn't matter - just feed it the same way in training and in inference.
If you train using train.py, most likely you still do need to translate to RGB.

EdjeElectronics, 2019-11-06:
I finally implemented the conversion from BGR to RGB! Thanks for the input.

closed 2019-11-06 by EdjeElectronics