EdjeElectronics/TensorFlow-Object-Detection-on-the-Raspberry-Pi #43: Headless rpi B+

alexharnozd, 2019-04-02:
Hi all.

Follow all the tuts and some workaround. 
Able to complete all tutorial with less error. 
But when i issue 

sudo pyhton3 object_detectionetcetc.py

I got this error
"cannot connect to X server"

Any work around on this?
Been struggling for 3 days with this.

Setup :
Pi camera
Rpi b+
My living room samsung tv via hdmi to rpi (headless) 

Did test take picture manually is ok
Did test take video manually is ok
Did test with keyboard usb with hdmi tv screen upon booth up

So far i m only SSH to rpi b+ via laptop(windows) & ssh via android phone.

gheesung, 2019-04-04:
If you have installed the default Raspian OS, you can enable VNC to your pi.

alexharnozd, 2019-04-04:
Hi @gheesung, thank you. 

I installed this ver Raspbian Stretch Lite from official rpi website.

Did enable VNC from raspi-config
And installed bunch of files in order to enable this vnc feature. 

Any other step should i make?

gheesung, 2019-04-04:
I have the normal Raspian Stretch installed and enable the VNC via raspi-config.

I did not notice if raspi-config installed additional files.

At the VNC client, I have disabled encryption to reduce the workload of the already over-worked pi.

lavanya5, 2020-04-23:
> Hi all.
>
> Follow all the tuts and some workaround.
> Able to complete all tutorial with less error.
> But when i issue
>
> sudo pyhton3 object_detectionetcetc.py
>
> I got this error
> "cannot connect to X [...]

hi alexharnozd,
Did you solved this issue?