EdjeElectronics/TensorFlow-Object-Detection-on-the-Raspberry-Pi #53: OpenCV install fails

technotrope, 2019-07-27:
(This is my first post, so please let me know if I should be following some other protocol.  Thanks.)

While following the guide (Many thanks for putting this guide together, BTW), I get the following error on the "pip3 install opencv-python" command:

  Could not find a version that satisfies the requirement opencv-python (from versions: )
No matching distribution found for opencv-python

Any suggestions as to a solution?

EdjeElectronics, 2019-08-03:
This happened to me too, and I think it's something to do with the latest OpenCV distribution in PyPi (pip). Try using the following command instead of the pip-based command. Let me know if it works!

```
sudo apt-get install python-opencv
```

technotrope, 2019-08-03:
Well, while following about 10 totally different blogs, I somehow managed to get it to work.  However, I can't say what finally did the trick.  There seems to be as many ways to do ANYTHING on the raspberry pi as there are people writing blogs!  LOL  Anyway, many thanks for your instructions.  

Now that I THINK I have everything installed, I plan to see if I can use your instructions to detect when pesky raccoons are frolicking in my koi pond.  Wish me luck, and many thanks.

EdjeElectronics, 2019-08-04:
Haha yes, there's definitely plenty of ways to skin a cat when it comes to the Raspberry Pi. I was facing this same issue when trying to get everything set up on my new Raspberry Pi 4. The issue was caused by the fact that the piwheels repository (which is where "pip3 install opencv-python" downloads the install package from) doesn't have a Python 3.7 version of OpenCV-Python. 

Raspberry Pi 4 only has Python 3.7, so it gives the "Could not find a version that satisfies the requirement opencv-python" error. I had to build OpenCV from source using [this guide](https://www.pyimagesearch.com/2018/09/26/install-opencv-4-on-your-raspberry-pi/).

closed 2019-08-04 by EdjeElectronics

technotrope, 2019-08-11:
Evan –

 

I’m terribly sorry if sending an email response to a closed incident email is a breach of protocol, but I couldn’t remember if you’d indicated what the best way is for asking questions about your videos.

 

I’m still slogging through various videos and textual instructions, trying to get my raspberry pi to detect objects.  Today’s problem is that I’ve reached the line:
 
                    protoc object_detection/protos/*.proto --python_out=.

 

and get a message that “protoc” is not a valid command.  I have Protobuf 3.6.1 installed, but cannot locate this command.  Would you have an idea as to where I’ve gone awry?

 

Thanks, in advance, for your help.

-Patrick Pirtle

 

Closed #53 <https://github.com/EdjeElectronics/TensorFlow-Object-Detection-on-the-Raspberry-Pi/issues/53> .