EdjeElectronics/TensorFlow-Object-Detection-on-the-Raspberry-Pi #62: Help wanted: Avoid having to clone ENTIRE tensorflow/models repository

EdjeElectronics, 2019-10-13:
When I originally wrote this guide, the tensorflow/models repository (https://github.com/tensorflow/models) was much smaller than it is today. The `git clone --recurse-submodules https://github.com/tensorflow/models.git` command in Step 5 downloaded about 200MB worth of data. Today, it's a 900MB download! The download takes too long for people who have poor internet connections.

Is there a way around downloading the full tensorflow/models repository? We can't only download the object_detection folder, because the Object Detection API uses files from the models/research and models/research/slim directories. It probably uses files from other places in the repository, too (I haven't checked to see all the modules the API uses).

I know I can just use TensorFlow Lite. I'm already working on a [guide showing how to set up TensorFlow Lite on the Pi](https://github.com/EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi).  However, I'd like to find a less data-hungry way to get the regular TensorFlow Object Detection API downloaded on the Pi.

How can we get the TensorFlow Object Detection API working without downloading the full tensorflow/models repository?

ladyada, 2019-10-13:
have you tried `--depth 1`?
https://linuxhint.com/git-shallow-clone-and-clone-depth/

EdjeElectronics, 2019-10-14:
@ladyada that looks promising, I will try it out. If it's a reasonably small size and everything works, I'll close this issue.

P.S. Thanks for the pull requests! SO awesome that we don't have to build protobuf-compiler from scratch anymore. 😸

ladyada, 2019-10-14:
awesome!

EdjeElectronics, 2019-10-17:
There's two changes that will make it a little better! 

1: Using `--depth 1` 
2: Removing the `--recurse-submodules` portion of the git command

I tested it out with these changes and it reduces the download size to about 370MB. And everything still works! I will update my guide with these changes. :+1:

closed 2019-10-17 by EdjeElectronics