EdjeElectronics/TensorFlow-Object-Detection-on-the-Raspberry-Pi #66: GraphDef

XxJoHnCeExX, 2019-11-05:
After installing eerything and running the Object_detection_picamera.py python script, I am getting this error:

Traceback (most recent call last):
  File "Object_detection_picamera.py", line 82, in <module>
    od_graph_def = tf.GraphDef()
AttributeError: module 'tensorflow' has no attribute 'GraphDef'

Any way of fixing this?

EdjeElectronics, 2019-11-07:
I'm pretty sure this is because you have TensorFlow v2.0 installed, and Object_detection_picamera.py isn't compatible with 2.0 yet. Could you try installing an older version of TensorFlow using

```
pip3 install tensorflow=1.13
```

and then try re-running the script? Please let me know if that works!

LefiS, 2020-06-05:
python3 Object_detection_picamera.py --usbcam
Traceback (most recent call last):
  File "Object_detection_picamera.py", line 82, in <module>
    od_graph_def = tf.GraphDef()
AttributeError: module 'tensorflow' has no attribute 'GraphDef'

pi@raspberrypi:~/tensorflow/models/research/object_detection $ **pip3 install tensorflow==1.13**
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting tensorflow==1.13
  Could not find a version that satisfies the requirement tensorflow==1.13 (from versions: 0.11.0, 1.12.0, 1.13.1, 1.14.0)
No matching distribution found for tensorflow==1.13
pi@raspberrypi:~/tensorflow/models/research/object_detection $

LefiS, 2020-06-05:
Ha I figured it out !
`Its pip3 install tensorflow==1.13.1`

1.13 is not enough :)