EdjeElectronics/TensorFlow-Object-Detection-on-the-Raspberry-Pi #67: ImportError: /usr/local/lib/python3.7/dist-packages/cv2/cv2.cpython-37m-arm-linux-gnueabihf.so: undefined symbol: __atomic_fetch_add_8

XinQing-Chaos, 2019-11-20:
Raspberry 4
python 3.7
tensorflow 2.0
i follow your guidence , and this issues happended at last ,
[image]
how should i do ?

mentianyi, 2019-11-21:
rp3b+,same error.

saifsmailbox98, 2019-11-21:
Same error using Raspbian Buster

saifsmailbox98, 2019-11-21:
```bash
$ LD_PRELOAD=/usr/lib/arm-linux-gnueabihf/libatomic.so.1.2.0 python3 Object_detection_picamera.py  --usbcam
```

mentianyi, 2019-11-22:
> Tried this workaround:
> 
> ```shell
> $ LD_PRELOAD=/usr/lib/arm-linux-gnueabihf/libatomic.so.1.2.0 python3 Object_detection_picamera.py  --usbcam
> ```
> 
> Getting a new error now:
> 
> ```
> Traceback (most recent call last):
>   File "Object_detection_picamera.py", line 27, in <module>
>     import tensorflow as tf
> ModuleNotFoundError: No module named 'tensorflow'
> ```
> 
> Not sure if they are relevant.
> 
> Edit: Not relevant. The workaround should work.

No help..

saifsmailbox98, 2019-11-22:
> No help..

Getting the same error?

Zach-OP, 2019-11-22:
I'm seeing this as well.

Zach-OP, 2019-11-22:
Hey, this is because the latest version of openCV doesn't work with RPi. Instead of pip3 install opencv-python try pip3 install opencv-python==3.4.6.27 
You will need to uninstall the latest version first. pip3 uninstall opencv-python

XinQing-Chaos, 2019-11-24:
i used another way to solve this issue yesterday, but it's  not  convenient 
i first install opencv follow other guidence  
https://www.pyimagesearch.com/2019/09/16/install-opencv-4-on-raspberry-pi-4-and-raspbian buster/  ,
then  i  skep  the  step  of  installing opencv  in this  guidence   . 
the result is  i need  to execute  some  extra code  everytime  i run the  "Object_detection_picamera.py",  they are  :
1.   sudo find / -type f -name ‘*atom*.so*’
2.   LD_PRELOAD=/usr/lib/arm-linux-gnueabihf/libatomic.so.1.2.0    python3   Object_detection_picamera.py

XinQing-Chaos, 2019-11-24:
i tried your way today ,  and it  did  works .  
thank   you  for  taking  the  time  to   solve  this  issue   @Zach-OP

jlbs86, 2019-12-02:
> Hey, this is because the latest version of openCV doesn't work with RPi. Instead of pip3 install opencv-python try pip3 install opencv-python==3.4.6.27
> You will need to uninstall the latest version first. pip3 uninstall opencv-python

thank you so much Zach-Op

ueray, 2019-12-18:
> 
> 
> Hey, this is because the latest version of openCV doesn't work with RPi. Instead of pip3 install opencv-python try pip3 install opencv-python==3.4.6.27
> You will need to uninstall the latest version first. pip3 uninstall opencv-python

When i try this, i get the error ''Could not install packages due to an EnvironmentError: 404 Client Error: Not found for url: https://pypi.org/simple/opencv-pyhton/''

Do you have any idea to achieve this problem?

PUPAN789ER, 2019-12-25:
> Hey, this is because the latest version of openCV doesn't work with RPi. Instead of pip3 install opencv-python try pip3 install opencv-python==3.4.6.27
> You will need to uninstall the latest version first. pip3 uninstall opencv-python

NICE !!!

insung3511, 2020-01-01:
> Hey, this is because the latest version of openCV doesn't work with RPi. Instead of pip3 install opencv-python try pip3 install opencv-python==3.4.6.27
> You will need to uninstall the latest version first. pip3 uninstall opencv-python

i have same error.. but is not helped. i try [this solution](http://www.yoyojacky.com/?p=559) and [this](https://github.com/piwheels/packages/issues/59#issuecomment-554525495). but also it doesnt work.. anybody plz help me 😭

ssspro, 2020-01-08:
> Hey, this is because the latest version of openCV doesn't work with RPi. Instead of pip3 install opencv-python try pip3 install opencv-python==3.4.6.27
> You will need to uninstall the latest version first. pip3 uninstall opencv-python

it did work with previous versions of opencv.. thanks

andromeda1998, 2020-01-27:
> Hey, this is because the latest version of openCV doesn't work with RPi. Instead of pip3 install opencv-python try pip3 install opencv-python==3.4.6.27
> You will need to uninstall the latest version first. pip3 uninstall opencv-python

Thank you so much! Worked like charm!

vietanha74, 2020-02-03:
Thanks you so much @Zach-OP  It really useful !!!

Riverlox12, 2020-02-09:
> Hey, this is because the latest version of openCV doesn't work with RPi. Instead of pip3 install opencv-python try pip3 install opencv-python==3.4.6.27
> You will need to uninstall the latest version first. pip3 uninstall opencv-python
It gives me this when i install the older version 

2020-02-09 05:51:42.552137: E tensorflow/core/platform/hadoop/hadoop_file_system.cc:132] HadoopFileSystem load error: libhdfs.so: cannot open shared object file: No such file or directory
mmal: mmal_vc_port_enable: failed to enable port vc.null_sink:in:0(OPQV): ENOSPC
mmal: mmal_port_enable: failed to enable connected port (vc.null_sink:in:0(OPQV))0x2948490 (ENOSPC)
mmal: mmal_connection_enable: output port couldn't be enabled
Traceback (most recent call last):
  File "Object_detection_picamera.py", line 124, in <module>
    camera = PiCamera()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 433, in __init__
    self._init_preview()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 513, in _init_preview
    self, self._camera.outputs[self.CAMERA_PREVIEW_PORT])
  File "/usr/lib/python3/dist-packages/picamera/renderers.py", line 558, in __init__
    self.renderer.inputs[0].connect(source).enable()
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 2212, in enable
    prefix="Failed to enable connection")
  File "/usr/lib/python3/dist-packages/picamera/exc.py", line 184, in mmal_check
    raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to enable connection: Out of resources

modarm-e, 2020-02-16:
> Hey, this is because the latest version of openCV doesn't work with RPi. Instead of pip3 install opencv-python try pip3 install opencv-python==3.4.6.27
> You will need to uninstall the latest version first. pip3 uninstall opencv-python

Thank you sooooooo much

gdoise, 2020-02-20:
> Hey, this is because the latest version of openCV doesn't work with RPi. Instead of pip3 install opencv-python try pip3 install opencv-python==3.4.6.27
> You will need to uninstall the latest version first. pip3 uninstall opencv-python

It worked for me on RPI 3 B.   It uninstalled my opencv_python-4.1.1.26 now it's ok with opencv-python==3.4.6.27

svsadafule, 2020-02-27:
> rp3b+,same error.

Use OpenCV contrib with the right version.

pip install opencv-contrib-python==4.1.0.25

bharath4284, 2020-02-27:
> Hey, this is because the latest version of openCV doesn't work with RPi. Instead of pip3 install opencv-python try pip3 install opencv-python==3.4.6.27
> You will need to uninstall the latest version first. pip3 uninstall opencv-python

i excecuted the same python 3.4.6.27 was installed but againg showing the same error

Riverlox12, 2020-02-28:
Okay thank you very much! Have a great day!

Edwiinrtz, 2020-03-07:
> Hey, this is because the latest version of openCV doesn't work with RPi. Instead of pip3 install opencv-python try pip3 install opencv-python==3.4.6.27
> You will need to uninstall the latest version first. pip3 uninstall opencv-python

it works well for me thanks

HarshBhatt09, 2020-03-27:
> > Hey, this is because the latest version of openCV doesn't work with RPi. Instead of pip3 install opencv-python try pip3 install opencv-python==3.4.6.27
> > You will need to uninstall the latest version first. pip3 uninstall opencv-python
> > It gives me this when i install the older version
> 
> 2020-02-09 05:51:42.552137: E tensorflow/core/platform/hadoop/hadoop_file_system.cc:132] HadoopFileSystem load error: libhdfs.so: cannot open shared object file: No such file or directory
> mmal: mmal_vc_port_enable: failed to enable port vc.null_sink:in:0(OPQV): ENOSPC
> mmal: mmal_port_enable: failed to enable connected port (vc.null_sink:in:0(OPQV))0x2948490 (ENOSPC)
> mmal: mmal_connection_enable: output port couldn't be enabled
> Traceback (most recent call last):
> File "Object_detection_picamera.py", line 124, in
> camera = PiCamera()
> File "/usr/lib/python3/dist-packages/picamera/camera.py", line 433, in **init**
> self._init_preview()
> File "/usr/lib/python3/dist-packages/picamera/camera.py", line 513, in _init_preview
> self, self._camera.outputs[self.CAMERA_PREVIEW_PORT])
> File "/usr/lib/python3/dist-packages/picamera/renderers.py", line 558, in **init**
> self.renderer.inputs[0].connect(source).enable()
> File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 2212, in enable
> prefix="Failed to enable connection")
> File "/usr/lib/python3/dist-packages/picamera/exc.py", line 184, in mmal_check
> raise PiCameraMMALError(status, prefix)
> picamera.exc.PiCameraMMALError: Failed to enable connection: Out of resources

Did you solve this? I have same issue

Martink-NZ, 2020-04-04:
I tried Zac's approach to uninstall and then install OpenCV. I tried using LD_PRELOAD. I tried using VIRTUALWRAPPER_PYTHON. I tried mkvirtualenv cv2. None of them worked. 
I use Pi 4, 4GB and Intel Movidius NCS2, Python 3.7. Any other suggestions are much appreciated. This has taken much too much time to get resolved. Hopefully there is an easy solution. 
Many thanks, Martin from NZ

Pupan-cpe, 2020-04-14:
> Hey, this is because the latest version of openCV doesn't work with RPi. Instead of pip3 install opencv-python try pip3 install opencv-python==3.4.6.27
> You will need to uninstall the latest version first. pip3 uninstall opencv-python

thanks

shubhamkhandhar, 2020-05-03:
I tried Zac's approach to uninstall and then install OpenCV. I tried using LD_PRELOAD. I tried using VIRTUALWRAPPER_PYTHON. I tried mkvirtualenv cv2. None of them worked.
I use Pi 4, 4GB and Intel Movidius NCS2, Python 3.7. Any other suggestions are much appreciated. This has taken much too much time to get resolved. Hopefully there is an easy solution.
Many thanks, Martin from NZ

I am also getting the same issue on rpi 4 . Someone plz help

shubhamkhandhar, 2020-05-03:
> I tried Zac's approach to uninstall and then install OpenCV. I tried using LD_PRELOAD. I tried using VIRTUALWRAPPER_PYTHON. I tried mkvirtualenv cv2. None of them worked.
> I use Pi 4, 4GB and Intel [...]

Did you found any solution please post it

60999, 2020-05-06:
$ vi .bashrc (or nano .bashrc  under root)
add：export LD_PRELOAD=/usr/lib/arm-linux-gnueabihf/libatomic.so.1 
exit with save file.
$ source .bashrc
$ python3
Python 3.7.3 (default, Dec 20 2019, 18:57:59) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import cv2
>>> 
It will work.

yoher, 2020-05-14:
Update following @Zach-OP advice to stay away from the latest version:
On a Pi4 today :
- OpenCV 4.1.1.26  KO
- OpenCV 4.1.0.25 imports properly and seems to run fine so far
=> Install with `pip3 install opencv-python==4.1.0.25`

shubhamkhandhar, 2020-05-15:
Dear Sir,
            I have tried both the ways you have suggested me, but it seems
none of them to be working, i am attaching the screeshots of the rpi
terminal,
od_problem_1) - It shows that the code doesn't run with sudo command.
od_problem_2) - It shows that the code does run without sudo command.
I am calling this code using another code which requires
administrative rights for working as it contains flask api to connect to
another api in another computer/laptop, so sudo needs to put there before
working. please help.

lukedavoli, 2020-05-15:
> 
> 
> Hey, this is because the latest version of openCV doesn't work with RPi. Instead of pip3 install opencv-python try pip3 install opencv-python==3.4.6.27
> You will need to uninstall the latest version first. pip3 uninstall opencv-python

god bless you

preyash2047, 2020-05-25:
> Hey, this is because the latest version of openCV doesn't work with RPi. Instead of pip3 install opencv-python try pip3 install opencv-python==3.4.6.27
> You will need to uninstall the latest version first. pip3 uninstall opencv-python

thank you bro, it worked for me

Keylem, 2020-06-08:
> $ vi .bashrc (or nano .bashrc under root)
> add：export LD_PRELOAD=/usr/lib/arm-linux-gnueabihf/libatomic.so.1
> exit with save file.
> $ source .bashrc
> $ python3
> Python 3.7.3 (default, Dec 20 2019, [...]

Thanks worked like charm with 4.1.1 😍

xlla, 2020-06-22:
how to export LD_PRELOAD=/usr/lib/arm-linux-gnueabihf/libatomic.so.1
for sudo command ? 
I want to use framebuffer over ssh, if invoke without sudo , it complain 
pygame.error: Unable to open a console terminal

mdjunaidahmed0123, 2020-06-28:
> i excecuted the same python 3.4.6.27 was installed but againg showing the same error

same here. please resolve this :(

mdjunaidahmed0123, 2020-06-28:
> i excecuted the same python 3.4.6.27 was installed but againg showing the same error

hi bharath4284
i too tried the same and same error i got again. did you install tensorflow ? if u did it then please tell me the command.

zimzamyoyoman, 2020-07-06:
> Hey, this is because the latest version of openCV doesn't work with RPi. Instead of pip3 install opencv-python try pip3 install opencv-python==3.4.6.27
> You will need to uninstall the latest version first. pip3 uninstall opencv-python
THANKS A TON! THIS WORKED!

wDove1, 2020-08-08:
This happens when I try uninstalling any ideas?

```
pi@raspberrypi:~ $ pip3 uninstall opencv-python
Uninstalling opencv-python-4.1.1.26:
  Would remove:
    /usr/local/lib/python3.7/dist-packages/cv2/*
    /usr/local/lib/python3.7/dist-packages/opencv_python-4.1.1.26.dist-info/*
Proceed (y/n)? y
Exception:
Traceback (most recent call last):
  File "/usr/lib/python3.7/shutil.py", line 563, in move
    os.rename(src, real_dst)
PermissionError: [Errno 13] Permission denied: '/usr/local/lib/python3.7/dist-packages/cv2/LICENSE-3RD-PARTY.txt' -> '/tmp/pip-uninstall-wr2v2c8q/usr/local/lib/python3.7/dist-packages/cv2/LICENSE-3RD-PARTY.txt'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/_internal/cli/base_command.py", line 143, in main
    status = self.run(options, args)
  File "/usr/lib/python3/dist-packages/pip/_internal/commands/uninstall.py", line 75, in run
    auto_confirm=options.yes, verbose=self.verbosity > 0,
  File "/usr/lib/python3/dist-packages/pip/_internal/req/req_install.py", line 683, in uninstall
    uninstalled_pathset.remove(auto_confirm, verbose)
  File "/usr/lib/python3/dist-packages/pip/_internal/req/req_uninstall.py", line 224, in remove
    renames(path, new_path)
  File "/usr/lib/python3/dist-packages/pip/_internal/utils/misc.py", line 280, in renames
    shutil.move(old, new)
  File "/usr/lib/python3.7/shutil.py", line 578, in move
    os.unlink(src)
PermissionError: [Errno 13] Permission denied: '/usr/local/lib/python3.7/dist-packages/cv2/LICENSE-3RD-PARTY.txt'
```

Rich2020, 2020-11-21:
> This happens when I try uninstalling any ideas?
>
> ```
> pi@raspberrypi:~ $ pip3 uninstall opencv-python
> Uninstalling opencv-python-4.1.1.26:
>   Would remove: [...]

It says "permission denied" - you forgot to use sudo: 

`pip3 uninstall opencv-python`

Getsethi, 2021-04-07:
ImportError                               Traceback (most recent call last)
<ipython-input-23-3ae6fc25c03f> in <module>()
      4 import pybind11
      5 import astropy
----> 6 import pygedm

1 frames
/usr/local/lib/python3.7/dist-packages/pygedm/ne2001_wrapper.py in <module>()
      3 from functools import wraps
      4 from astropy import units as u
----> 5 from . import dmdsm     # f2py FORTRAN object
      6 from . import density   # f2py FORTRAN object
      7 

ImportError: /usr/local/lib/python3.7/dist-packages/pygedm/dmdsm.so: undefined symbol: PyCObject_Type

Please help resolve this

DorineMaillet, 2021-05-03:
> Update following @Zach-OP advice to stay away from the latest version:
> On a Pi4 today :
> 
> * OpenCV 4.1.1.26  KO
> * OpenCV 4.1.0.25 imports properly and seems to run fine so far
>   => Install with `pip3 install opencv-python==4.1.0.25`

Thanks a lot, it still works today and I needed it soooo much !!!!