EdjeElectronics/TensorFlow-Object-Detection-on-the-Raspberry-Pi #75: E tensorflow/core/platform/hadoop/hadoop_file_system.cc:132] HadoopFileSystem load error: libhdfs.so: cannot open shared object file: No such file or directory

shABanty, 2020-03-01:
I'm getting this error while running picamera.py
Anyone know how to resolve?
Pls help, thanks!
[image]

tazelisk, 2020-03-19:
Same issue here....

tazelisk, 2020-03-19:
> I'm getting this error while running picamera.py
> Anyone know how to resolve?
> Pls help, thanks!
> [image]

jklancnik, 2020-03-30:
I also have this problem. Latest Raspbian Buster, Raspberry pi 3b+, last compatible version of Open CV installed (3.4.6.27)

Anyone solved this one?

santoshmohan, 2020-04-11:
I have the same problem as well

domingosohan, 2020-08-07:
I am having the same issue. Do we have to install tensorflow and openCv in the same folder ?

gtimk6, 2021-06-15:
same problem here. anyone help?

TuruuDo, 2021-12-04:
Yeah, I have very similar problem. Please, someone teach.