EdjeElectronics/TensorFlow-Object-Detection-on-the-Raspberry-Pi #76: HadoopFileSystem load error

rubinho101, 2020-03-01:
Hello, while running Object_detection_picamera.py I get the following issue, please help. Thx

` tensorflow/core/platform/hadoop/hadoop_file_system.cc:132] HadoopFileSystem load error: libhdfs.so: cannot open shared object file: No such file or directory
WARNING:tensorflow:From /home/pi/tensorflow1/models/research/object_detection/utils/label_map_util.py:138: The name tf.gfile.GFile is deprecated. Please use tf.io.gfile.GFile instead.

WARNING:tensorflow:From Object_detection_picamera.py:82: The name tf.GraphDef is deprecated. Please use tf.compat.v1.GraphDef instead.

WARNING:tensorflow:From Object_detection_picamera.py:88: The name tf.Session is deprecated. Please use tf.compat.v1.Session instead.

mmal: mmal_vc_port_enable: failed to enable port vc.null_sink:in:0(OPQV): ENOSPC
mmal: mmal_port_enable: failed to enable connected port (vc.null_sink:in:0(OPQV))0x397c320 (ENOSPC)
mmal: mmal_connection_enable: output port couldn't be enabled
Traceback (most recent call last):
  File "Object_detection_picamera.py", line 124, in <module>
    camera = PiCamera()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 433, in __init__
    self._init_preview()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 513, in _init_preview
    self, self._camera.outputs[self.CAMERA_PREVIEW_PORT])
  File "/usr/lib/python3/dist-packages/picamera/renderers.py", line 558, in __init__
    self.renderer.inputs[0].connect(source).enable()
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 2212, in enable
    prefix="Failed to enable connection")
  File "/usr/lib/python3/dist-packages/picamera/exc.py", line 184, in mmal_check
    raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to enable connection: Out of resources`

rubinho101, 2020-03-01:
HadoopFileSystem load error: libhdfs.so: cannot open shared object file: No such file or directory
WARNING:tensorflow:From /home/pi/tensorflow1/models/research/object_detection/utils/label_map_util.py:138: The name tf.gfile.GFile is deprecated. Please use tf.io.gfile.GFile instead.

WARNING:tensorflow:From Object_detection_picamera.py:82: The name tf.GraphDef is deprecated. Please use tf.compat.v1.GraphDef instead.

WARNING:tensorflow:From Object_detection_picamera.py:88: The name tf.Session is deprecated. Please use tf.compat.v1.Session instead.

: cannot connect to X server

yoyojacky, 2020-03-08:
DISPLAY=:0.0 run your code here.

g2721, 2020-04-13:
> HadoopFileSystem load error: libhdfs.so: cannot open shared object file: No such file or directory
> WARNING:tensorflow:From [...]

Same problem

marcospiridon, 2020-06-15:
Same problem

g2721, 2020-06-15:
2 possible solutions first: you install raspbinian with gui or just install X (basically the gui), just look for raspi x server install

g2721, 2020-06-15:
This issue can be closed now since it isn’t a bug