EdjeElectronics/TensorFlow-Object-Detection-on-the-Raspberry-Pi #78: undefined symbol: __atomic_fetch_add_8

johanf82, 2020-03-18:
Hi,

I did your tutorial 3 times, but I keep getting the same error:

Traceback (most recent call last):
  File "Object_detection_picamera.py", line 23, in <module>
    import cv2
  File "/usr/local/lib/python3.7/dist-packages/cv2/__init__.py", line 3, in <module>
    from .cv2 import *
ImportError: /usr/local/lib/python3.7/dist-packages/cv2/cv2.cpython-37m-arm-linux-gnueabihf.so: undefined symbol: __atomic_fetch_add_8

Did I do something wrong?

Thanks!

g2721, 2020-04-13:
No same for me, but I found out you can use LD_PRELOAD=/usr/lib/arm-linux-gnueabihf/libatomic.so.1 python3 Object_detection_picamera.py

g2721, 2020-04-13:
But then I got this here: 2020-04-13 14:03:06.554427: E tensorflow/core/platform/hadoop/hadoop_file_system.cc:132] HadoopFileSystem load error: libhdfs.so: cannot open shared object file: No such file or directory
WARNING:tensorflow:From /home/pi/tensorflow1/models/research/object_detection/utils/label_map_util.py:138: The name tf.gfile.GFile is deprecated. Please use tf.io.gfile.GFile instead.

WARNING:tensorflow:From Object_detection_picamera.py:82: The name tf.GraphDef is deprecated. Please use tf.compat.v1.GraphDef instead.

WARNING:tensorflow:From Object_detection_picamera.py:88: The name tf.Session is deprecated. Please use tf.compat.v1.Session instead.

: cannot connect to X server

lavanya5, 2020-04-23:
> But then I got this here: 2020-04-13 14:03:06.554427: E tensorflow/core/platform/hadoop/hadoop_file_system.cc:132] HadoopFileSystem load error: libhdfs.so: cannot open shared object file: No such [...]

hi g2721,
I am facing the same problem.Did you find any solution for this error?
Thank you.

g2721, 2020-04-23:
@lavanya5 Yeah i got it working. I guess you are running it on the console ? For me installing Pixel (the gui for raspberrys or the complete Gui) worked

g2721, 2020-04-23:
Just look for install x server on pi

lavanya5, 2020-04-23:
> Just look for install x server on pi and type startX or so to activate the gui

Thank you