EdjeElectronics/TensorFlow-Object-Detection-on-the-Raspberry-Pi #79: TypeError: __init__() missing 1 required positional argument: 'name'

HarshBhatt09, 2020-03-27:
Hey, I have followed your installation steps for setting up object detection with Raspberry Pi. I have installed tensorflow 2.1 and opencv3.4.6.27.
I have done following changes in the Object_detection code
`tf.compat.v1.GraphDef()   # -> instead of tf.GraphDef()
tf.compat.v2.io.gfile.GFile()   # -> instead of tf.gfile.GFile()`

But now I am getting error "TypeError: __init__() missing 1 required positional argument: 'name'" when I run following command in Terminal
`python3 Object_detection_picamera.py `

Can anyone provide any solution for this? Do I need to change anything more in Object Detection Code or this is an error of any library or dependencies?

shashwat1225, 2020-05-20:
I've been facing a similar issue. Has anyone got it yet?

DmitryBerdnik, 2022-05-06:
Same issue here