EdjeElectronics/TensorFlow-Object-Detection-on-the-Raspberry-Pi #82: Issue With starting Project import cv2

peteronksley, 2020-04-14:
Hi 
Thank you for you tutorial 

i am having a issue starting the project at
python3 Object_detection_picamera.py --usbcam
i get this message bellow which says cv2 is missing i tried looking for the missing files using ld and ls but nothing is missing

any ideas? 

Traceback (most recent call last):
  File "Object_detection_picamera.py", line 23, in <module>
    import cv2
  File "/usr/local/lib/python3.7/dist-packages/cv2/__init__.py", line 3, in <module>
    from .cv2 import *
ImportError: /usr/local/lib/python3.7/dist-packages/cv2/cv2.cpython-37m-arm-linux-gnueabihf.so: undefined symbol: __atomic_fetch_add_8

LutBit, 2020-04-28:
have the same problem. with this command its still working `LD_PRELOAD=/usr/lib/arm-linux-gnueabihf/libatomic.so.1 python3 Object_detection_picamera.py` found in another post.
Still seems to be just a workaround.

vivekjha832002, 2020-05-25:
helo  facing issue while starting 

this error is coming
Traceback (most recent call last):
  File "Object_detection_picamera.py", line 23, in <module>
    import cv2
  File "/usr/local/lib/python3.7/dist-packages/cv2/__init__.py", line 3, in <module>
    from .cv2 import *
ImportError: /usr/local/lib/python3.7/dist-packages/cv2/cv2.cpython-37m-arm-linux-gnueabihf.so: undefined symbol: __atomic_fetch_add_8

mali-x, 2020-06-02:
How could I solve this problem that I don't need to add at startup: `LD_PRELOAD=/usr/lib/arm-linux-gnueabihf/libatomic.so.1 python3 Object_detection_picamera.py`