EdjeElectronics/TensorFlow-Object-Detection-on-the-Raspberry-Pi #83: No matching distribution found for tensorflow

sohartma, 2020-04-15:
Hi,

I'm trying to install tensorflow on my RasPi3 b+. 
I've already updated and upgraded.

pip3 install tensorflow
pip install tensorflow
python2 -m pip install tensorflow
python3 -m pip install tensorflow

All give me this error:
  Could not find a version that satisfies the requirement tensorflow (from versions: )
No matching distribution found for tensorflow

How to fix this error?

bcm1001, 2020-05-03:
Hi, i have same problem
2020-05-03 14:07:40.471816: E tensorflow/core/platform/hadoop/hadoop_file_system.cc:132] HadoopFileSystem load error: libhdfs.so: cannot open shared object file: No such file or directory

0xNoor, 2022-12-22:
Is there any update on this?