EdjeElectronics/TensorFlow-Object-Detection-on-the-Raspberry-Pi #88: issue with cv2.imshow()

adsenger, 2020-08-26:
I'm getting this error running the script:

```
Traceback (most recent call last):
  File "Object_detection_picamera.py", line 161, in <module>
    cv2.imshow('Object detector', frame)
cv2.error: OpenCV(4.4.0) /tmp/pip-install-fx8ry_hd/opencv-python/opencv/modules/highgui/src/window.cpp:651: error: (-2:Unspecified error) The function is not implemented. Rebuild the library with Windows, GTK+ 2.x or Cocoa support. If you are on Ubuntu or Debian, install libgtk2.0-dev and pkg-config, then re-run cmake or configure script in function 'cvShowImage'
```

I installed opencv using `pip install opencv-python` after installing all the dependencies, not sure what was missing.

woutervanranst, 2020-09-24:
same issue here

woutervanranst, 2020-09-26:
I apt get installed the 2 packages, did an u install of opencv and reinstall opencv an it worked from there