EdjeElectronics/TensorFlow-Object-Detection-on-the-Raspberry-Pi #94: Trying to install OpenCV, but sudo apt-get update doesn't help.

gabriel-bronfman, 2021-11-20:
Hi,

Really want this tutorial to work. I am unable to install OpenCV. When running sudo apt-get install qt4-dev-tools libatlas-base-dev and sudo apt-get install libavcodec-dev libavformat-dev libswscale-dev libv4l-dev, I get an error where I am unable to fetch some archives. It tells me to do the sudo apt-get update, but it doesn't help. 

What do I do?

starkfire, 2023-04-07:
If your Raspberry Pi runs Debian 11 (bullseye), you might need to add the old Raspbian repository in `/etc/apt/sources.list`:

```
deb http://archive.raspbian.org/raspbian stretch main contrib non-free
```

Then do a `sudo apt-get update` afterwards before installing the packages again.

drelvenkee, 2023-11-04:
try do this command "pip install mediapipe" - it works for me.