EdjeElectronics/TensorFlow-Object-Detection-on-the-Raspberry-Pi #97: OpenCV `ImportError: libIlmImf-2_2.so.23: cannot open shared object file: No such file or directory` error

MWimbish1988, 2022-01-30:
# ImportError: libIlmImf-2_2.so.23: cannot open shared object file: No such file or directory

Hello, 

I am using python3.7 Raspberry Pi 3 B+ and I an trying install opencv to start a project that is a Self-Driving Robotic Car by Use of Computer Vision. The only library I am having trouble with is OpenCV. I first attempt `sudo apt-get install python3-opencv`. It worked. However, I need it for python3.7. It installs to python 3.9. I then remove it and try `pip3.7 install opencv-python`. That didn't work. For some reason the version 4.5.5.62 is not installing. So, I was specific by typing `pip3.7 install opencv-python==4.5.3.56` because I read it is compatible with python 3.7. This version installs; however, I recieved the following message: 

    (driving_cv) pi@raspberrypi:~ $ python3
    Python 3.7.4 (default, Jan 29 2022, 23:25:05) 
    [GCC 10.2.1 20210110] on linux
    Type "help", "copyright", "credits" or "license" for more information.
    >>> import cv2
    Traceback (most recent call last):
    File "<stdin>", line 1, in <module>
    File "/home/pi/.local/lib/python3.7/site-packages/cv2/__init__.py", line 5, in <module>
        from .cv2 import *
    ImportError: libIlmImf-2_2.so.23: cannot open shared object file: No such file or directory
    >>> 

I did some research and read I should try the follow command. `sudo apt install libopenexr23`. However, when executing, it says to replace with `libopenexr25` which I do. I recheck `python3.7 -c "import cv2"` again and get the same error.

Can someoneassist with this issue, please?

AmiirGh, 2022-07-24:
I have the same error.

drelvenkee, 2023-11-04:
try this fix "pip install mediapipe" - it works for me

skeephonyeah, 2024-10-24:
this doesn't work for me. any other work around available?